rtic-rs/rtic #715: Whacky behavior with `rtic_monotonics::rp2040::Timer`

DusterTheFirst, 2023-03-26:
When using the `rtic_monotonics::rp2040::Timer` monotonic timer implementation on a raspberry pi pico, concurrent timers will appear to "lag", firing delayed and quickly trying to catch up. This behavior can be seen in the two videos below.  Swapping the timer implementation for `rtic_monotonics::systick::Systick` (also while only having one of the cargo features on at a time) will produce the expected, non-"laggy" output.

Video with Systick timer

Video with rp2040 timer

I have not looked deeply into the code behind the rp2040 timer, but if I were to guess at what is happening, it looks like the timer interrupt (alarm) is being set for the longest delay in the future, rather than the shortest. As short delays will seem to be blocked on longer ones that are happening concurrently. This happens with multiple tasks with different priorities, or one task and `futures_util::future::join`.

DusterTheFirst, 2023-03-29:
I have created a minimal reproduction of the bug, which you can find here:

https://github.com/DusterTheFirst/pico-things/blob/7666ad5cf011400a034bbea10990545a89ecb30d/projects/pico-rtic-timer-2040-repr/src/main.rs#L89-L128

<details>
<summary>Photo of the output with <code>Systick</code> timer (expected result)</summary>

[image]

</details>

<details>
<summary>Photo of the output with <code>rp2040</code> timer (odd behavior)</summary>

[image]

</details>

You can see in the above images, looking at the timestamp in the defmt logs, that in the rp2040 example the second and third "One Second" messages are delayed until the firing of the "Three Seconds" message. This can be confirmed in real-life, watching the logs become more jumpy. This is *not* a problem with the timestamps.

datdenkikniet, 2023-04-05:
The link you posted doesn't work :(

DusterTheFirst, 2023-04-06:
@datdenkikniet Oops, it looks like that repository was private. I have made it public.

korken89, 2023-04-07:
Thanks for reporting! I'll reproduce and fix it

korken89, 2023-04-08:
There is a fix PR up! Thank to @datdenkikniet for debugging!

commit 2fb09d4a (2023-04-09): Merge #727

727: Fix if a enqueued instant is first in a non-empty queue r=perlindgren a=korken89

Fixes #715
PR #727 (2023-04-09): Fix if a enqueued instant is first in a non-empty queue
Fixes #715
closed 2023-04-09 by bors[bot]