rtic-rs/rtic #717: Adding a dispatcher argument to app causes build error

deltronix, 2023-03-29:
In the process of creating a usb midi driver for the STM32H743 I ran into this build error when adding a software task and corresponding dispatcher:

The code builds and runs fine without the software task and dispatcher. It looks to me like there is some kind of interaction between the usb-device crate and rtic?
```
error[E0107]: this type alias takes 1 generic argument but 2 generic arguments were supplied
  --> src/main.rs:8:1
   |
8  | #[app(device = stm32h7xx_hal::pac, peripherals = true, dispatchers = [EXTI0])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   | |
   | expected 1 generic argument
   | help: remove this generic argument
   |
note: type alias defined here, with 1 generic parameter: `T`
  --> /home/douwe/.cargo/registry/src/github.com-1ecc6299db9ec823/usb-device-0.2.9/src/lib.rs:95:10
   |
95 | pub type Result<T> = core::result::Result<T, UsbError>;
   |          ^^^^^^ -
   = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)
```

korken89, 2023-04-07:
Hi,

Can you share the code for debugging?

korken89, 2023-04-15:
Hi, please reopen this is you are able to provide an example that shows the error.
Sadly I've not been able to reproduce it.

closed 2023-04-15 by korken89