rtic-rs/rtic #740: spawn_{at,after} keeps block until time has reached

vDorst, 2023-04-20:
I am working on a Modbus project on a stm32F411 blackpill.
I want to scheduled a task to generate a timeout or reschedule if a new byte is received.
But I do this in a uart interrupt.

I noticed that the ` spawn_after`   and `spawn_at` keeps blocked until time has reached.
Code is here https://github.com/vDorst/stm32_modbus/blob/main/src/main.rs#L707

probe-run output:
```
tmr create
└─ stm32_modbus::app::uart2_handle::{closure#0} @ src/main.rs:705
--- Here is a 10sec delay ---
tmr create done
└─ stm32_modbus::app::uart2_handle::{closure#0} @ src/main.rs:708
MB: 1 InTrans
└─ stm32_modbus::modbus::{impl#0}::char_recv @ src/modbus.rs:53
tmr timeout
└─ stm32_modbus::app::tmr_modbus @ src/main.rs:719
```

Is this normal behavior?
Or do I something wrong?

Versions: 

`cargo 1.68.2 (6feb7c9cf 2023-03-26)`

I am fully up-to-date.
```
stm32_modbus v0.1.0 (/home/r/dev/rust/stm32_modbus)
├── cortex-m v0.7.7
│   ├── bare-metal v0.2.5
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   ├── bitfield v0.13.2
│   ├── critical-section v1.1.1
│   ├── embedded-hal v0.2.7
│   │   ├── nb v0.1.3
│   │   │   └── nb v1.1.0
│   │   └── void v1.0.2
│   └── volatile-register v0.2.1
│       └── vcell v0.1.3
├── cortex-m-rt v0.7.3
│   └── cortex-m-rt-macros v0.7.0 (proc-macro)
│       ├── proc-macro2 v1.0.56
│       │   └── unicode-ident v1.0.8
│       ├── quote v1.0.26
│       │   └── proc-macro2 v1.0.56 (*)
│       └── syn v1.0.109
│           ├── proc-macro2 v1.0.56 (*)
│           ├── quote v1.0.26 (*)
│           └── unicode-ident v1.0.8
├── cortex-m-rtic v1.1.4
│   ├── bare-metal v1.0.0
│   ├── cortex-m v0.7.7 (*)
│   ├── cortex-m-rtic-macros v1.1.6 (proc-macro)
│   │   ├── proc-macro-error v1.0.4
│   │   │   ├── proc-macro-error-attr v1.0.4 (proc-macro)
│   │   │   │   ├── proc-macro2 v1.0.56 (*)
│   │   │   │   └── quote v1.0.26 (*)
│   │   │   │   [build-dependencies]
│   │   │   │   └── version_check v0.9.4
│   │   │   ├── proc-macro2 v1.0.56 (*)
│   │   │   ├── quote v1.0.26 (*)
│   │   │   └── syn v1.0.109 (*)
│   │   │   [build-dependencies]
│   │   │   └── version_check v0.9.4
│   │   ├── proc-macro2 v1.0.56 (*)
│   │   ├── quote v1.0.26 (*)
│   │   ├── rtic-syntax v1.0.3
│   │   │   ├── indexmap v1.9.3
│   │   │   │   └── hashbrown v0.12.3
│   │   │   │   [build-dependencies]
│   │   │   │   └── autocfg v1.1.0
│   │   │   ├── proc-macro2 v1.0.56 (*)
│   │   │   ├── quote v1.0.26 (*)
│   │   │   └── syn v1.0.109 (*)
│   │   └── syn v1.0.109 (*)
│   ├── heapless v0.7.16
│   │   ├── defmt v0.3.4
│   │   │   ├── bitflags v1.3.2
│   │   │   └── defmt-macros v0.3.4 (proc-macro)
│   │   │       ├── defmt-parser v0.3.2
│   │   │       │   └── thiserror v1.0.40
│   │   │       │       └── thiserror-impl v1.0.40 (proc-macro)
│   │   │       │           ├── proc-macro2 v1.0.56 (*)
│   │   │       │           ├── quote v1.0.26 (*)
│   │   │       │           └── syn v2.0.15
│   │   │       │               ├── proc-macro2 v1.0.56 (*)
│   │   │       │               ├── quote v1.0.26 (*)
│   │   │       │               └── unicode-ident v1.0.8
│   │   │       ├── proc-macro-error v1.0.4 (*)
│   │   │       ├── proc-macro2 v1.0.56 (*)
│   │   │       ├── quote v1.0.26 (*)
│   │   │       └── syn v1.0.109 (*)
│   │   ├── hash32 v0.2.1
│   │   │   └── byteorder v1.4.3
│   │   └── stable_deref_trait v1.2.0
│   │   [build-dependencies]
│   │   └── rustc_version v0.4.0
│   │       └── semver v1.0.17
│   ├── rtic-core v1.0.0
│   └── rtic-monotonic v1.0.0
│   [build-dependencies]
│   └── version_check v0.9.4
├── defmt v0.3.4 (*)
├── defmt-rtt v0.4.0
│   ├── critical-section v1.1.1
│   └── defmt v0.3.4 (*)
├── embedded-graphics v0.7.1
│   ├── az v1.2.1
│   ├── byteorder v1.4.3
│   ├── embedded-graphics-core v0.3.3
│   │   ├── az v1.2.1
│   │   └── byteorder v1.4.3
│   ├── float-cmp v0.8.0
│   │   └── num-traits v0.2.15
│   │       [build-dependencies]
│   │       └── autocfg v1.1.0
│   └── micromath v1.1.1
├── embedded-hal v0.2.7 (*)
├── fixedvec v0.2.4
├── heapless v0.7.16 (*)
├── log v0.4.17
│   └── cfg-if v1.0.0
├── nb v1.1.0
├── panic-probe v0.3.1
│   ├── cortex-m v0.7.7 (*)
│   └── defmt v0.3.4 (*)
├── rmodbus v0.7.3
│   ├── heapless v0.7.16 (*)
│   └── ieee754 v0.2.6
├── ssd1306 v0.7.1
│   ├── display-interface v0.4.1
│   ├── display-interface-i2c v0.4.0
│   │   ├── display-interface v0.4.1
│   │   └── embedded-hal v0.2.7 (*)
│   ├── display-interface-spi v0.4.1
│   │   ├── byte-slice-cast v0.3.5
│   │   ├── display-interface v0.4.1
│   │   └── embedded-hal v0.2.7 (*)
│   ├── embedded-graphics-core v0.3.3 (*)
│   └── embedded-hal v0.2.7 (*)
├── stm32f4xx-hal v0.15.0
│   ├── bare-metal v1.0.0
│   ├── bitflags v1.3.2
│   ├── cortex-m v0.7.7 (*)
│   ├── cortex-m-rt v0.7.3 (*)
│   ├── embedded-dma v0.2.0
│   │   └── stable_deref_trait v1.2.0
│   ├── embedded-hal v0.2.7 (*)
│   ├── embedded-hal v1.0.0-alpha.8
│   │   └── nb v1.1.0
│   ├── embedded-storage v0.2.0
│   ├── fugit v0.3.6
│   │   └── gcd v2.3.0
│   ├── fugit-timer v0.1.3
│   │   ├── fugit v0.3.6 (*)
│   │   └── nb v1.1.0
│   ├── nb v1.1.0
│   ├── rand_core v0.6.4
│   ├── stm32f4 v0.15.1
│   │   ├── bare-metal v1.0.0
│   │   ├── cortex-m v0.7.7 (*)
│   │   ├── cortex-m-rt v0.7.3 (*)
│   │   └── vcell v0.1.3
│   ├── time v0.3.20
│   │   └── time-core v0.1.0
│   └── void v1.0.2
├── systick-monotonic v1.0.1
│   ├── cortex-m v0.7.7 (*)
│   ├── fugit v0.3.6 (*)
│   └── rtic-monotonic v1.0.0
├── w5500-dhcp v0.4.2
│   ├── defmt v0.3.4 (*)
│   └── w5500-hl v0.9.0
│       ├── defmt v0.3.4 (*)
│       └── w5500-ll v0.10.4
│           ├── defmt v0.3.4 (*)
│           └── embedded-hal v0.2.7 (*)
└── w5500-mqtt v0.1.2
    ├── defmt v0.3.4 (*)
    └── w5500-hl v0.9.0 (*)
```

burrbull, 2023-04-21:
Task priorities

vDorst, 2023-04-21:
Although it did work, assigning a priority to `Systick` interrupt which is lower than the `UART` interrupt.

I run the same code on nrf52840 and that did work without assign a priority to `systick`.
Only `uart2` interrupt had a priority = 2 assigned.

So for me it feels like that there is still something not right.
My exception is that the schedule code programs the capture register of systick timer and then returns back to the caller.
But not wait until the time is reached.

datdenkikniet, 2023-04-22:
This really shouldn't block, so it's a bit weird that it does.

I would definitely recommend that you decrease the frequency of your `SysTick`. Running it at 100 KHz is highly discouraged, as this causes a `SysTick` interrupt to occur every `sysclk/100_000` cycles (which seems to be 160 in this project). Please try that and see if anything changes.

korken89, 2023-04-22:
A quick look tells me the issue or probably your Systick setup, it's ticking at 100kHz.

Reduce this, probably to 1kHz, and our should work fine.
Systick monotonic is really only for simple testing, above 1 kHz it'll use significant resources. :)

vDorst, 2023-04-23:
Ah yes I see and understand what I am doing and what is wrong.
Reducing the systick did help of course.
My assumption, without looking as the arm docs, was that the systick worked like a capture compare timer.
Programs a capture register that only generate a interrupt at that point.
It seems what this is better suited for me is already in the HAL: https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/timer/monotonic.rs

Thanks for the help!

closed 2023-04-23 by vDorst