rtic-rs/rtic #749: Monotonic trait confusion

diggit, 2023-05-12:
Hi,
I'd like to implement custom monotonic using TIM3 on STM32H7. Ideally using HAL rather than PAC.
I am unsure which `Monotonic` trait to implement.
There is one in rtic-time (this - rtic repo) and another one in rtic-monotonic repo.
All the existing implementations I found (except those in rtic-monotonics in this repo) are implementing trait from rtic-monotonic repo.
Why there are 2 different definitions of same trait?

Please bear with me, I am total rust and rtic beginner.

My goal is to implement something like this for STM32H7
https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/timer/monotonic.rs

burrbull, 2023-05-13:
I think you should implement both.
`rtic-monotonic` one is used in RTIC 1.x which works on stable.
`rtic-tick` is optimized for using with `async` an required by future RTIC 2.0

burrbull, 2023-05-14:
Added extended 64-bit `rtic-monotonic` timer in f4xx-hal
https://github.com/stm32-rs/stm32f4xx-hal/pull/640

datdenkikniet, 2023-05-27:
I second the advice from burrbull. 

The traits are quite similar, so it should not be very difficult to do a single implementation and `#[cfg]` out the parts that aren't shared.

closed 2023-05-27 by datdenkikniet