rtic-rs/rtic #762: you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml cargo check only

haata, 2023-06-04:
After switching to rtic v2.0.0 recently, I've seen this odd error indicating haven't enabled the rt feature in my pac crate (I have)

```bash
[main]: cargo make check
[cargo-make] INFO - cargo make 0.36.8
[cargo-make] INFO - Calling cargo metadata to extract project info
[cargo-make] INFO - Cargo metadata done
[cargo-make] INFO - Project: keystonefs
[cargo-make] INFO - Build File: Makefile.toml
[cargo-make] INFO - Task: check
[cargo-make] INFO - Profile: development
[cargo-make] INFO - Running Task: legacy-migration
[cargo-make] INFO - Execute Command: "cargo" "check"
    Checking keystonefs v0.1.1 (/home/hyatt/Source/kiibohd/kiibohd-firmware/inputclub/keyboards/keystone/fullsize)
error[E0412]: cannot find type `Peripherals` in module `rtic::export`
  --> inputclub/keyboards/keystone/fullsize/src/bin.rs:43:1
   |
43 | #[app(device = kiibohd_atsam4s::hal::pac, peripherals = true, dispatchers = [UART1, USART0, USART1, SSC, PWM, ACC, TWI0, TWI1])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `rtic::export`
   |
   = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this struct
   |
45 +     use crate::app::you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::Peripherals;
   |

error[E0425]: cannot find function `run` in module `rtic::export`
  --> inputclub/keyboards/keystone/fullsize/src/bin.rs:43:1
   |
43 | #[app(device = kiibohd_atsam4s::hal::pac, peripherals = true, dispatchers = [UART1, USART0, USART1, SSC, PWM, ACC, TWI0, TWI1])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `rtic::export`
   |
   = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0425]: cannot find function `lock` in module `rtic::export`
  --> inputclub/keyboards/keystone/fullsize/src/bin.rs:43:1
   |
43 | #[app(device = kiibohd_atsam4s::hal::pac, peripherals = true, dispatchers = [UART1, USART0, USART1, SSC, PWM, ACC, TWI0, TWI1])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `rtic::export`
   |
   = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `interrupt` in `export`
  --> inputclub/keyboards/keystone/fullsize/src/bin.rs:43:1
   |
43 | #[app(device = kiibohd_atsam4s::hal::pac, peripherals = true, dispatchers = [UART1, USART0, USART1, SSC, PWM, ACC, TWI0, TWI1])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ could not find `interrupt` in `export`
   |
   = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this enum
   |
45 +     use crate::app::you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt;
   |

error[E0433]: failed to resolve: could not find `Peripherals` in `export`
  --> inputclub/keyboards/keystone/fullsize/src/bin.rs:43:1
   |
43 | #[app(device = kiibohd_atsam4s::hal::pac, peripherals = true, dispatchers = [UART1, USART0, USART1, SSC, PWM, ACC, TWI0, TWI1])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ could not find `Peripherals` in `export`
   |
   = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this struct
   |
45 +     use crate::app::you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::Peripherals;
   |

error[E0425]: cannot find function `nop` in module `rtic::export`
  --> inputclub/keyboards/keystone/fullsize/src/bin.rs:43:1
   |
43 | #[app(device = kiibohd_atsam4s::hal::pac, peripherals = true, dispatchers = [UART1, USART0, USART1, SSC, PWM, ACC, TWI0, TWI1])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `rtic::export`
   |
   = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)

Some errors have detailed explanations: E0412, E0425, E0433.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `keystonefs` (bin "keystonefs") due to 6 previous errors
[cargo-make] ERROR - Error while executing command, exit code: 101
[cargo-make] WARN - Build Failed.
```

`cargo make build` and `cargo make clippy` work fine, so there must be something I'm missing that's not obvious.

The code is available here: https://github.com/kiibohd/kiibohd-firmware/tree/main/inputclub/keyboards/keystone/fullsize

AfoHT, 2023-06-07:
I tried to recreate the issue, but sadly I did not manage to build 1f2a16a7c00028ff319fa014e17e2b85f5c9bfc4 using `cargo make check` from within the `keyboards/keystone/fullsize` folder.

Had to add a `todo!()` in hidio.rs to not fail compilation: 
[image]

With that sorted, keystonefs still fails unfortunately: 
[image]

Feels like I'm clearly doing something wrong, could it be you have something in your Cargo.lock?
Nightly used was `rustc 1.72.0-nightly (b2b34bd83 2023-06-06)  ` 

The `rt` feature error can be quite misleading, but due to the macro error reporting shortcomings it is what it is. I had a look into your dependencies and that seems to all be in order regarding the `rt` feature, via `atsam4s8b` feature.

Could you investigate target/rtic-expansion.rs?

Maybe provide some further steps/hints how to recreate the issue?

haata, 2023-06-08:
Thank you so much for looking! (and my apologies for the problems so far...).

This file is part of a git sub-module (I should add a better error message for that...).

```bash
git submodule update --init
```

```bash
 --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', /home/hyatt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/layouts-rs-0.1.0/src/lib.rs:58:46
  stack backtrace:
     0:     0x55ee9ea51951 - std::backtrace_rs::backtrace::libunwind::trace::hf9423e7319e7b273
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
     1:     0x55ee9ea51951 - std::backtrace_rs::backtrace::trace_unsynchronized::h66acda1b30f743d6
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
     2:     0x55ee9ea51951 - std::sys_common::backtrace::_print_fmt::hd68a2e46e11b4b1a
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/sys_common/backtrace.rs:65:5
     3:     0x55ee9ea51951 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h9597144efc95fd0d
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/sys_common/backtrace.rs:44:22
     4:     0x55ee9ea751df - core::fmt::rt::Argument::fmt::h78292bc3d727b7ea
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/core/src/fmt/rt.rs:138:9
     5:     0x55ee9ea751df - core::fmt::write::h3c0df26dbae30f71
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/core/src/fmt/mod.rs:1094:21
     6:     0x55ee9ea4f4c1 - std::io::Write::write_fmt::h9885b87fb4cdbd4b
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/io/mod.rs:1713:15
     7:     0x55ee9ea51765 - std::sys_common::backtrace::_print::h8dd7723bf85d46b9
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/sys_common/backtrace.rs:47:5
     8:     0x55ee9ea51765 - std::sys_common::backtrace::print::h4d1dcd147b97d0e0
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/sys_common/backtrace.rs:34:9
     9:     0x55ee9ea52c07 - std::panicking::default_hook::{{closure}}::h77fbf8c391b9b7dc
    10:     0x55ee9ea529f4 - std::panicking::default_hook::h8cf79dcd3fda8d66
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/panicking.rs:288:9
    11:     0x55ee9ea530bc - std::panicking::rust_panic_with_hook::h499030a402682852
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/panicking.rs:705:13
    12:     0x55ee9ea52fb7 - std::panicking::begin_panic_handler::{{closure}}::h6ee14f69855f4395
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/panicking.rs:597:13
    13:     0x55ee9ea51d86 - std::sys_common::backtrace::__rust_end_short_backtrace::he15120d3b7461309
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/sys_common/backtrace.rs:151:18
    14:     0x55ee9ea52d02 - rust_begin_unwind
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/std/src/panicking.rs:593:5
    15:     0x55ee9e60cec3 - core::panicking::panic_fmt::hc8e9b6dda341d3de
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/core/src/panicking.rs:67:14
    16:     0x55ee9e60d453 - core::result::unwrap_failed::h24db349c24a6622b
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/core/src/result.rs:1651:5
    17:     0x55ee9e8708ac - core::result::Result<T,E>::unwrap::hfc9336571bd36187
                                 at /rustc/dd5d7c729d4e8a59708df64002e09dbcbc4005ba/library/core/src/result.rs:1076:23
    18:     0x55ee9e860e49 - layouts_rs::Layout::from_file::hdb3f602f4992c9e5
                                 at /home/hyatt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/layouts-rs-0.1.0/src/lib.rs:58:20
    19:     0x55ee9e860fba - layouts_rs::LayoutSource::fetch_layout::h979a0ae9e6834ac5
                                 at /home/hyatt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/layouts-rs-0.1.0/src/lib.rs:110:45
    20:     0x55ee9e861377 - layouts_rs::Layouts::get_layout::h12c604b00b6209ff
                                 at /home/hyatt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/layouts-rs-0.1.0/src/lib.rs:153:26
    21:     0x55ee9e849f25 - kll_compiler::types::Action::kll_core_condition::h5ab786f09db9dee4
                                 at /home/hyatt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/kll-compiler-0.1.3/src/types.rs:846:22
    22:     0x55ee9e846ecf - kll_compiler::types::ResultList::kll_core_guide::hc84c471b252bcb0f
                                 at /home/hyatt/.cargo/registry/src/index.crates.io-6f17d22bba15001f/kll-compiler-0.1.3/src/types.rs:184:43
    23:     0x55ee9e755d31 - kll_compiler::emitters::kllcore::KllCoreData::new::h749ee1eb24fafb0a
[... cut ...]
[cargo-make] INFO - Task: build
[cargo-make] INFO - Profile: development
[cargo-make] INFO - Running Task: legacy-migration
[cargo-make] INFO - Execute Command: "cargo" "build" "--target" "thumbv7em-none-eabi"
   Compiling keystonefs v0.1.1 (/home/hyatt/git/kiibohd-firmware/inputclub/keyboards/keystone/fullsize)
    Finished dev [optimized + debuginfo] target(s) in 1.59s
[cargo-make] INFO - Build Done in 2.00 seconds.
```

I'll take a closer look at `target/rtic-expansion.rs` to see what find.

Thanks!

AfoHT, 2023-06-14:
I completely missed there were submodules. Thank you for the gist, I ran cargo format on it, to make it a bit nicer on the eyes: https://gist.github.com/AfoHT/ab60e9e78c52596f8db3c836c7d9ea31

If you look at the top, there's 

```
#[doc = r" The RTIC application module"]
pub mod app {
    #[doc = r" Always include the device crate which contains the vector table"]
    use kiibohd_atsam4s::hal::pac as you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml;
```
which leads us to that every time we see this interesting macro workaround of a long name, it is referring to your PAC

Later on used as 
```
        let mut core: rtic::export::Peripherals = rtic::export::Peripherals::steal().into();
        let _ = you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt::UART1;
        let _ = you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt::USART0;
        let _ = you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt::USART1;
        let _ = you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt::SSC;
        let _ = you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt::PWM;
        let _ = you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt::ACC;
        let _ = you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt::TWI0;
        let _ = you_must_enable_the_rt_feature_for_the_pac_in_your_cargo_toml::interrupt::TWI1;
```
which seems fine, I think I saw this exact path in your PAC, given the RT feature

No obvious red flags to me, would it be feasible to try to slim down the example complexity to try to find a MVE?

AfoHT, 2023-06-28:
Had another look at this, and it seems it was a bit simpler than I think we anticipated it should be... I hope you can confirm this over at your end. Got the submodule and building became much easier :tada: :smile: 

Created a PR, mentioned above ^

[image]

haata, 2023-06-29:
Thanks so much!

To explain a bit more on this, my project uses some std-only (as in nostd incompatible) crates that are used with build.rs. This means each build target is split between thumbv7em-none-eabi  and x86_64-unknown-linux-gnu. This can make things tricky to build with some tools that assume one or the other.
This mode is enabled in cargo using
```toml
[unstable]
features = ["host_dep"]
```

cargo udeps likes to have --all-targets set, but there seem to be issues with this atm (fortunately, cargo-make takes care of each of the targets so it's not a problem).

Previously I didn't need to use `cargo check --target thumbv7em-none-eabi` which is where the confusion started.

Anyways, thanks so much!

closed 2023-06-29 by haata