rtic-rs/rtic #777: "item constrains opaque type that is not in its signature" for rtic 2.0.0

cnmozzie, 2023-07-12:
I am trying to run the stm32f3_blinky example in Windows. However, after I change my rust toolchain to nightly, there will be the following error:

error: item constrains opaque type that is not in its signature
  --> src\main.rs:14:1
   |
14 | #[app(device = stm32f3xx_hal::pac, peripherals = true, dispatchers = [SPI1])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: this item must mention the opaque type in its signature in order to be able to register hidden types
note: this item must mention the opaque type in its signature in order to be able to register hidden types
  --> src\main.rs:14:1
   |
14 | #[app(device = stm32f3xx_hal::pac, peripherals = true, dispatchers = [SPI1])]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)

error: could not compile `stm32f3-blinky` (bin "stm32f3-blinky") due to previous error

dgoodlad, 2023-07-12:
I just hit this same issue; while I haven't had time to track it down to a specific commit or feature in nightly, I did confirm that `nightly-2023-07-08` _works_ while `nightly-2023-07-09` is broken.

AfoHT, 2023-07-14:
This issue was discussed in depth during this week's meeting. @korken89 will investigate further 

Start of discussion: 
https://matrix.to/#/!yafYEipFNsXDdwiHMT:matrix.org/$FuzFTd8VcVaMtzNr2F8j8Ni3whBEjWtf27ym1zM9Yj0?via=matrix.org&via=kde.org&via=tchncs.de

PR #782 (2023-07-25): Fixed new TAIT requirement and release v2.0.1 of RTIC
Closes #777
closed 2023-07-25 by korken89