rtic-rs/rtic #778: Access to shared resource requires and does not require mutable declaration at the same time

90degs2infty, 2023-07-13:
I need some help on understanding when to access shared resources in a `let mut` way and when a mere `let` (without the `mut`) is sufficient, please. Can someone explain to me why in below example in `timer1`, I'm required to declare `bar` as `let mut bar = ctx.shared.bar` while at the same time `let bar = ctx.shared.bar` is sufficient in `timer2`?

From how I read the example in [the docs](https://rtic.rs/2/book/en/by-example/resources.html#shared-resources-and-lock), the `mut` shouldn't be necessary in `timer1`, or am I getting things utterly wrong?

Thanks a lot! 🙏 

Below example is based on [the app template](https://github.com/rtic-rs/defmt-app-template). Please find a self-contained, compileable setup [here](https://github.com/90degs2infty/rtic-mut-not-mut).

```rust
#![no_main]
#![no_std]
#![feature(type_alias_impl_trait)]

use test_app as _; // global logger + panicking-behavior + memory layout

#[rtic::app(
    device = nrf52840_hal::pac,
    dispatchers = [SWI0_EGU0]
)]
mod app {

    // Shared resources go here
    #[shared]
    struct Shared {
        foo: u8,
        bar: u8,
    }

    // Local resources go here
    #[local]
    struct Local {}

    #[init]
    fn init(_cx: init::Context) -> (Shared, Local) {
        (Shared { foo: 42, bar: 43 }, Local {})
    }

    // Optional idle, can be removed if not needed.
    #[idle]
    fn idle(_: idle::Context) -> ! {
        defmt::info!("idle");

        #[allow(clippy::empty_loop)]
        loop {}
    }

    #[task(binds = TIMER1, shared = [bar])]
    fn timer1(ctx: timer1::Context) {
        // Why do I need mut here?
        // let bar = ctx.shared.bar; // triggers a 'cannot borrow `bar` as mutable' below
        let mut bar = ctx.shared.bar;
        bar.lock(|_bar| {});

        // The same is true for
        // ctx.shared.bar.lock(|_bar| { });
    }

    #[task(binds = TIMER2, shared = [foo, bar])]
    fn timer2(ctx: timer2::Context) {
        #[allow(clippy::disallowed_names)]
        let foo = ctx.shared.foo;
        let bar = ctx.shared.bar;

        (bar, foo).lock(|_bar, _foo| {});
    }
}
```

Just in case it matters:

```sh
> rustc --version
rustc 1.73.0-nightly (33a2c2487 2023-07-12)
```

korken89, 2023-07-25:
Hi,

This comes from the move of the `foo` and `bar` into the tuple `(foo, bar)`.
As this move happens this tuple is "upgraded" to mutable as it is a temporary, and then the mutable requirement from `lock` is upheld.

Hope that helps!

closed 2023-07-27 by korken89

90degs2infty, 2023-07-27:
Yep, that helps. Thank you very much, @korken89 ! I wasn't aware of the temporary being mutable in itself - good to know 😅 

One more thing I noticed re-reading [the docs](https://rtic.rs/2/book/en/by-example/resources.html#shared-resources-and-lock) (which I leave here just in case somebody else stumbles upon this): in the docs `foo`'s context is declared mutable ( `async fn foo(mut c: foo::Context)`) whereas in the example I stated above the context is declared immutable (e.g. `fn timer1(ctx: timer1::Context)`). In consequence, the call `c.shared.shared.lock()` is possible in the docs but in above example a move to some `mut` variable (or a temporary as you pointed out) is required.