rtic-rs/rtic #780: assertion failed: !self.0.access(cs).freeq.is_empty() in rtic-sync 1.0.1 when trying to send data to channel

pistolero, 2023-07-24:
> panicked at 'assertion failed: !self.0.access(cs).freeq.is_empty()', /home/serg/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rtic-sync-1.0.1/src/channel.rs:301:17

I've tried checking the code myself, but it is not trivial, so any help would be appreciated

korken89, 2023-07-25:
Hi, thanks for the report!

Do you have an example we can use to reproduce this? 
I checked tests and this assert is quite well tested so I'm not sure what's wrong, any help to narrow it down would be appreciated!

pistolero, 2023-07-26:
I was hitting it consistently for some time and now it's gone.

I'll try to find a case where it is reproducing consistently and show you how to reproduce it. I think it was related to publishing messages before reader started to read them.

korken89, 2023-07-26:
Thanks, I hope you can trigger it again!

korken89, 2023-08-05:
Hi @pistolero I just wanted to check-in if you ran into this again?

I have a theory on what can happen, and would like to check if this is similar to what you did.
If you at a higher priority use `try_send` and at the same time use `send` at a lower priority I think the free queue can become inconsistent if hit just right.
I've not been able to reproduce it in tests though...

Were you doing something similar to this? I've tested extensively with different dropping/forgetting of the receiver and were unfortunately not able to reproduce :(

pistolero, 2023-08-05:
Not yet, had no chance to get back to this 😓

I think it was happening when queue was full, I had a software thread which was posting data to queue periodically and this assertion was failing when receiver of the queue were blocked (there were no Ethernet connection).

commit 41c861ba (2025-03-12): Fix #780
commit 0a01fde4 (2025-03-12): Fix #780
closed 2025-03-12 by korken89