rtic-rs/rtic #800: Question: `Send` shared resource container for `thumbv6m-none-eabi` target

smmoosavi, 2023-09-11:
Hi,

I want to define a shareable mutable container like `Rc<RefCell>` and store that in the `#[shared]` struct. But `Rc<RefCell>` is not `Send`.

> Types of `#[shared]` resources have to be `Send`.

Then I tried to use the `Arbiter` struct. It's not `Clone` so I have to wrap it in `Arc`. `Arc<Arbiter>` is working as expected. but `Arc` is not Implemented for the `thumbv6m-none-eabi` target.

```rust
use alloc::sync::Arc;
//         ^^^^ could not find `sync` in `alloc`
```

I also tried `&Arbiter`, but it forced me to have a lifetime for my struct. which cause lifetime error when used by Mutex:

```rust
assert_eq!(mtx.lock(|option| option.wait_for_value()).await, 1);
//                   ------- ^^^^^^^^^^^^^^^^^^^^^^^ returning this value requires that `'1` must outlive `'2`
//                   |     |
//                   |     return type of closure is async_option::wait_for_value_future::WaitForValueFuture<'2, i32>
//                   has type `&'1 mut async_option::AsyncOption<i32>`
```

What is the best alternative for the `Rc<RefCell>` for the `thumbv6m-none-eabi` target?

I created a sample [repo] with multiple implementations.

[repo]: https://github.com/smmoosavi/rtic-async-shared-resource/

korken89, 2023-09-13:
Hi,

The only way to have non-`Send` is in a resource in an priority = 0 task.
If you want to store something (where the inner value is `Send`), `Arbiter` is a good choice.
I'm not sure I understand your usecase good enough to give any better advice.

Hope that helps!

smmoosavi, 2023-09-18:
> I'm not sure I understand your usecase good enough to give any better advice.

We need access to the same `Arbiter` from multiple tasks. The `Arc` struct was the perfect solution for us, But it does not exist in `thumbv6m-none-eabi`

korken89, 2023-09-20:
Hi,

The `rtic_sync::arbiter::Arbiter` should work from multiple tasks, it is `Sync`.
So either share it via a `static` or an immutable resource would be my recommendation.
`Arc` is not needed for `Arbiter`.

AfoHT, 2023-10-04:
Did you figure this out by using the Arbiter?

smmoosavi, 2023-10-07:
@AfoHT 

No, @korken89's recommendation is more about how to avoid the problem. Sometimes I have a non-static struct that needs one simple field and one shared field, so it can't be either static or immutable

I add the [`ValueContainer`][vc-src] example to the repo

[vc-src]: https://github.com/smmoosavi/rtic-async-shared-resource/blob/main/src/value_container.rs

smmoosavi, 2023-10-29:
I found a [portable-atomic-util] crate that provides the `Arc` type for
`thumbv6m-none-eabi`. It is part of [portable atomic][portable-atomic-git] and uses [portable-atomic] crate internally. rtic use
[atomic-polyfill] which is deprecated and encourages the use [portable-atomic]
instead.

[portable-atomic-util]: https://crates.io/crates/portable-atomic-util
[portable-atomic]: https://crates.io/crates/portable-atomic
[atomic-polyfill]: https://crates.io/crates/atomic-polyfill
[portable-atomic-git]: https://github.com/taiki-e/portable-atomic

closed 2025-03-12 by korken89