rtic-rs/rtic #807: Strategy Design Pattern and RTIC

hanusek, 2023-09-23:
Hello!
I am a newbie in Embedded Rust and RTIC framework.
I would like to implement a design pattern - [Strategy](https://refactoring.guru/design-patterns/strategy/rust/example) or classic pointer to the function.

I created this trait:
```rust
trait OperationStrategy 
{
    fn operate(&mut self);
}
```
and main struct 
```rust
struct Card<T: OperationStrategy> {
    strategy: T,
}

impl<T: OperationStrategy> Card<T> {
    pub fn new(strategy: T) -> Self {
        Self { strategy }
    }

    pub fn operate(&mut self) {
        self.strategy.operate();
    }
}
```

Rust Playground:
https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=3a24cfddcbbdb9f150209686c606e7ec

Embedded Rust (`no_std`) hasn't `Box`, `Arc` and `Mutext.
I have a problem with adding the struct `Card` to `Shared`. 
```    
#[shared]
struct Shared {}
```
How to fix it? 
Please Help

My platform: STM32F103

hanusek, 2023-09-23:
I implemented two solutions.
https://github.com/hanusek/rtic-strategy-pattern/tree/solution1-enum
https://github.com/hanusek/rtic-strategy-pattern/tree/solution2-fnptr
What do you think?

AfoHT, 2023-10-04:
If these two solutions work for you, great! We discussed this briefly during previous week's meeting, and some opinions came out of it, but nobody had tried this pattern so not much to draw from it. 

If there is something else to do here, please reopen

closed 2023-10-04 by AfoHT