rtic-rs/rtic #839: Link errors when running rtic-monotonics example

jcard0na, 2023-11-17:
Hi,

I'm trying to run [this example code](https://rtic.rs/dev/book/en/by-example/delay.html#tasks-with-delay) from the book on my STM32L072C8Tx board.

For this, I have configured the required crates as follow:
```
$ cat Cargo.toml  | grep rtic-mono
rtic = { version = "2.0.1", features = ["cortex-m", "thumbv6-backend", "rtic-monotonics"] }
rtic-monotonics = { version = "1.2.0", features = ["stm32l052c8", "cortex-m-systick"] }
```
This is the complete code from the example with minimal modifications, essentially just replacing `semihosting` with `rtt`:

```rust
#![no_main]
#![no_std]
#![feature(type_alias_impl_trait)]

use panic_rtt_target as _;

#[rtic::app(device = stm32l0xx_hal::pac, dispatchers = [SPI1])]
mod app {
    use rtic_monotonics::systick::*;
    use rtt_target::{rprintln, rtt_init_print};

    #[shared]
    struct Shared {}

    #[local]
    struct Local {}

    #[init]
    fn init(cx: init::Context) -> (Shared, Local) {
        rtt_init_print!();
        rprintln!("init");

        let systick_token = rtic_monotonics::create_systick_token!();
        Systick::start(cx.core.SYST, 12_000_000, systick_token);

        foo::spawn().ok();
        bar::spawn().ok();
        baz::spawn().ok();

        (Shared {}, Local {})
    }

    #[task]
    async fn foo(_cx: foo::Context) {
        rprintln!("hello from foo");
        Systick::delay(100.millis()).await;
        rprintln!("bye from foo");
    }

    #[task]
    async fn bar(_cx: bar::Context) {
        rprintln!("hello from bar");
        Systick::delay(200.millis()).await;
        rprintln!("bye from bar");
    }

    #[task]
    async fn baz(_cx: baz::Context) {
        rprintln!("hello from baz");
        Systick::delay(300.millis()).await;
        rprintln!("bye from baz");
    }
}
```

The code compiles but fails to link due to undefined symbols, which are interrupt names.

```
  = note: rust-lld: error: undefined symbol: RCC
          >>> referenced by rtic_playground.9c29a361ea81e178-cgu.0
          >>>               /home/javier/dev/rtic-playground/target/thumbv6m-none-eabi/release/deps/rtic_playground-00b34e20d939bb11.rtic_playground.9c29a361ea81e178-cgu.0.rcgu.o:(__INTERRUPTS)
          >>> did you mean: RTC
          
          rust-lld: error: undefined symbol: DMA1_CHANNEL4_7
          >>> referenced by rtic_playground.9c29a361ea81e178-cgu.0
          >>>               /home/javier/dev/rtic-playground/target/thumbv6m-none-eabi/release/deps/rtic_playground-00b34e20d939bb11.rtic_playground.9c29a361ea81e178-cgu.0.rcgu.o:(__INTERRUPTS)
          
          rust-lld: error: undefined symbol: ADC_COMP
          >>> referenced by rtic_playground.9c29a361ea81e178-cgu.0
          >>>               /home/javier/dev/rtic-playground/target/thumbv6m-none-eabi/release/deps/rtic_playground-00b34e20d939bb11.rtic_playground.9c29a361ea81e178-cgu.0.rcgu.o:(__INTERRUPTS)
          
          rust-lld: error: undefined symbol: USART4_USART5
          >>> referenced by rtic_playground.9c29a361ea81e178-cgu.0
          >>>               /home/javier/dev/rtic-playground/target/thumbv6m-none-eabi/release/deps/rtic_playground-00b34e20d939bb11.rtic_playground.9c29a361ea81e178-cgu.0.rcgu.o:(__INTERRUPTS)
          
          rust-lld: error: undefined symbol: TIM3
          >>> referenced by rtic_playground.9c29a361ea81e178-cgu.0
          >>>               /home/javier/dev/rtic-playground/target/thumbv6m-none-eabi/release/deps/rtic_playground-00b34e20d939bb11.rtic_playground.9c29a361ea81e178-cgu.0.rcgu.o:(__INTERRUPTS)
          
          rust-lld: error: undefined symbol: TIM7
          >>> referenced by rtic_playground.9c29a361ea81e178-cgu.0
          >>>               /home/javier/dev/rtic-playground/target/thumbv6m-none-eabi/release/deps/rtic_playground-00b34e20d939bb11.rtic_playground.9c29a361ea81e178-cgu.0.rcgu.o:(__INTERRUPTS)
          
          rust-lld: error: undefined symbol: I2C3
          >>> referenced by rtic_playground.9c29a361ea81e178-cgu.0
          >>>               /home/javier/dev/rtic-playground/target/thumbv6m-none-eabi/release/deps/rtic_playground-00b34e20d939bb11.rtic_playground.9c29a361ea81e178-cgu.0.rcgu.o:(__INTERRUPTS)
          
          rust-lld: error: undefined symbol: AES_RNG_LPUART1
          >>> referenced by rtic_playground.9c29a361ea81e178-cgu.0
          >>>               /home/javier/dev/rtic-playground/target/thumbv6m-none-eabi/release/deps/rtic_playground-00b34e20d939bb11.rtic_playground.9c29a361ea81e178-cgu.0.rcgu.o:(__INTERRUPTS)
```
This looks like a possible misconfiguration, but so far I have not been able to track it down.  All other examples from the book that don't use `rtic-monotonics` compile and run successfully.

Can anyone point me to any documentation explaining how to configure the `rtic-monotonics` trait?  I enabled `stm32l052c8,cortex-m-systick` because they made sense to me, but could not find them documented anywhere.

korken89, 2023-11-22:
Hi, we did a large rework of the STM32 monotonics to fix a lot of issues. Could you try with `rtic-monotonics v1.3.0` and check if the issue persists?
If it does can you put the project on GH?

jcard0na, 2023-11-23:
Oh, nice.  Yes `v1.3.0` solves this problem and I can run the book example.
Thank you very much for your good work!  :hugs:

closed 2023-11-23 by jcard0na