rtic-rs/rtic #859: Higher resolution monotonic timer (on stm32)

deltronix, 2023-12-06:
It seems the resolution of timer based monotonics on the stm32 is a hard limit set by this constant:
`const TIMER_HZ: u32 = 1_000_000;`

Is there a reason for this, or could TIMER_HZ be made a generic? I would like to have a bit more resolution for my use case.

burrbull, 2023-12-06:
The issue is in async executor, which require queue to be 'static' variable. But 'static' variables don't support any generics.

Finomnis, 2023-12-06:
@deltronix Are you sure that this would really be helpful for your usecase? I had the impression that scheduling overhead and similar already has a noise of multiple ticks, but maybe I'm wrong

Finomnis, 2023-12-06:
We can of course at least provide a higher resolution alternative that can be enabled via feature, like we have for the `systick` monotonic. What resolution do you have in mind?

Finomnis, 2023-12-07:
@burrbull I agree the same problem on imxrt the other way round. The monotonic we have can take any clock source but it's fixed to 1MHz; it would be useful if the user could use the low power dock in the khz range with it. Otherwise we will have to provide a low power version of it at some point.

Finomnis, 2023-12-07:
Another solution would be to move all of the monotonic code into macros that the user has to instantiate somewhere.
Yet another solution would be to avoid the static nature of the thing all together, and actually instantiate a monotonic object somewhere. It would work with the interrupts just fine if we generated the static memory inside of the interrupt token macros.

All of this would require a major rework, though, and a v2 of both the rtic-monotonics and the rtic-time crate.

Finomnis, 2023-12-07:
So I did some measurements. I used the following code on an nrf52840 on a 1MHz monotonic:
```rust
#[task(local = [led])]
async fn blink(cx: blink::Context) {
    let blink::LocalResources { led, .. } = cx.local;

    let mut next_tick = Mono::now();
    let mut blink_on = false;
    loop {
        blink_on = !blink_on;
        if blink_on {
            led.set_high().unwrap();
        } else {
            led.set_low().unwrap();
        }

        next_tick += 1.micros();
        Mono::delay_until(next_tick).await;
    }
}
```

And got the following waveform:
[image]

As you can see, it's nowhere close to 1MHz, so increasing precision is kind of pointless. Everything with a higher frequency might have to be generated via dedicated PWM timer.

Be aware that the reason this waveform is at the wrong frequency but still stable is because scheduling actually never happens, every time `Mono::delay` is called it returns immediately, because the given instant is already in the past.

If you try with `20.micros()`, you can see that `.await`ing happens sometimes, but not always, causing a very glitchy pattern:

[image]

This is when I try with `50.micros()`. It is about the highest I managed to achieve on nrf52840 without starting to get glitches:

[image]

There might be chips where higher frequencies are stable, like the imxrt1060 family or high-performance stm32 chips, but I doubt that a scheduling precision beyond `1us` would actually be stable on those either.

Finomnis, 2023-12-07:
Ah, for reading the time. I get that usecase. Sadly it would require a major rework

Finomnis, 2024-02-28:
Should get solved by #874, once merged. This makes monotonic tick rates no longer hard coded.

closed 2024-04-10 by AfoHT