rtic-rs/rtic #875: Clean application shutdown

Eugeny, 2023-12-29:
If there a way to cleanly shutdown a RTIC app, cancel (or wait for) any pending task futures, release resources and release the Peripherals struct back to user code?

The use case for this is jumping back to bootloader code on STM32 for USB DFU, which requires all peripherals to be deinitialized first.

perlindgren, 2024-01-17:
RTIC does not currently support mode changes. You may programmatically encode a shutdown, and then manually perform a soft reset. Then in the `init` task you can check the reset condition, to confirm that the reset was not a cold boot. Dependent on platform, there can ways to store additional information from previous boot state if wanted.

perlindgren, 2024-01-17:
We close this issue, please feel free to re-open if there is any question.

closed 2024-01-17 by perlindgren