rtic-rs/rtic #899: nested lock() can allow access to unlocked resources

dyoung-smg, 2024-03-06:
The current interrupt masking approach in cortex_source_mask.rs breaks in some cases when resources are heavily shared between tasks and locks are nested. 

Minimal example (haven't checked for compile, sorry). Adapted from examples, issue was discovered on a system being developed on an RP2040.

```
#![no_main]
#![no_std]
#![deny(warnings)]
#![deny(unsafe_code)]
#![deny(missing_docs)]
#![feature(type_alias_impl_trait)]

use panic_semihosting as _;

#[rtic::app(device = lm3s6965, dispatchers = [GPIOA])]
mod app {
    use cortex_m_semihosting::{debug, hprintln};

    #[shared]
    struct Shared {
        shared1: u32, //shared between foo and bar, masks GPIOA and GPIOB
        shared2: u32, //same
    }

    #[local]
    struct Local {}

    #[init]
    fn init(_: init::Context) -> (Shared, Local) {
        foo::spawn().unwrap();

        (Shared { shared: 0 }, Local {})
    }

    #[task(shared = [shared1, shared2])]
    async fn foo(mut c: foo::Context) {
        let shared1 = c.shared.shared1;
        let shared2 = c.shared.shared2;
        shared2.lock(|shared2| {
            //GPIOA and GPIOB masked
           shared2 += 1 //OK access to shared 1
            shared1.lock(|shared1| {
            //GPIOA and GPIOB masked again
                      shared1 += 1; //so far so ggood
            });
            //GPIOA and GPIOB unmasked
            //racy access to shared2!
            *shared2 += 1;
        });
        //GPIOA and GPIOB unmasked again
    }

    #[task(binds = GPIOB, priority = 2, shared = [shared1, shared2])]
    fn bar(mut c: bar::Context) {
        c.shared.shared2.lock(|shared2| {
            *shared2 += 1; // this can race with access in foo!
        });
    }}
}```

dyoung-smg, 2024-03-12:
I believe this is on the latest version, but I think the changes in that file are not related to the issue we are seeing (I think my example may have been confusing, as I constructed it from the example in the documentation which uses a different chip). The issue we are seeing in practice is on the RP2040 which uses this implementation of lock: https://github.com/rtic-rs/rtic/blob/0b365f03eb77302663b751305aac7641b2721eb3/rtic/src/export/cortex_source_mask.rs#L122-L147 , not the one changed in that patch

korken89, 2024-03-13:
Hi,

This is now fixed in #902 and awaiting merge & release.

closed 2024-03-13 by AfoHT

dyoung-smg, 2024-03-14:
Awesome, thanks for the fix!

PR #902 (2024-03-13): Fix thumbv6 source masking
We unconditionally enabled interrupts on exit of locks, now we only enable interrupts that were disabled by the mask.

Closes #899