rtic-rs/rtic #921: rtic_monotonics 2.0 not in crates.io???

tristanguerragomez, 2024-04-16:
Hi,

Please forgive me if I'm off, just started transitioning into the rtic world.

All the examples point to rtic_monotonics 2.0, but that's not available on crates.io?  Tried with 1.5 but there seems to have been pretty major breaking changes on that end.

Am I missing something?

Rene

AfoHT, 2024-04-16:
Hi!

You are correct, currently rtic-monotonics 2.0 is not yet released to crates.io
When building rtic-monotonics and friends from git master branch you can use the latest syntax, even before it is released

Alternatively, look at the examples prior to #874 : https://github.com/rtic-rs/rtic/tree/e4cc5fd17b8a2df332af0ee25c8bd7092e66afb0 and use the current non-latest-bleeding edge released RTIC :)

tristanguerragomez, 2024-04-18:
Thank you for pointing that out @AfoHT , I'm starting to wrap my head around the current state of affairs...

I'm guessing rtic-monotonics 2.0 hasn't yet been release due to stability/documentation concerns (trust me, I get it, ensuring stability in the embedded world is brutally difficult, and who likes writing documentation?).  What I'm wondering is why not release Alpha/Beta versions?  Like Linus said, "given enough eyeballs, all bugs are shallow".  My thinking is that with Alpha/Beta versions 1) more "eyeballs" will be on the latest code base which is really where you want the Dev efforts to be focused on and 2) crate consumers are made aware of stability concerns.

Seems to me like a low effort effort way to get the code to be easily tested by the community (doing a PoC building rtic-monotonics and friends from git master branch is fine, but doing so with multiple Devs and CI gets very ugly very quickly...  I'm sure you understand).

Food for thought

alecrivers, 2024-04-19:
Just a +1 that I found this very confusing. Also, it seems like there exists

[rtic-monotonics on crates.io](https://crates.io/crates/rtic-monotonics): version 1.5.0.
[rtic-monotonic on crates.io](https://crates.io/crates/rtic-monotonic): version 1.0.0.
rtic-monotonics inside the rtic crate: version 2.0.0.

Furthermore, I was confused by the fact that there's
[cortex-m-rtic on crates.io](https://crates.io/crates/cortex-m-rtic): This is version 1.1.4, and contains the message "IMPORTANT: This crate is published as cortex-m-rtic on crates.io but the name of the library is rtic" and has no mention of being deprecated
[rtic on crates.io](https://crates.io/crates/rtic): This is the one we're supposed to be using, I take it?

The examples I found on the web are also mixed in terms of what version of things they point to. And as a minor point, I also noticed that RTIC's documentation refers to [defmt-app-template](https://github.com/rtic-rs/defmt-app-template), but that has one clone a repo called "app-template" -- though that appears to be just a redirect to defmt-app-template.

Overall, it was pretty confusing trying to get set up in the RTIC ecosystem. But I think I'm almost at blinky :)

tristanguerragomez, 2024-04-19:
FYI: The `stm32f411_rtc_interrupt` example is also off, it references a yanked version of rtic that has an [rtic-monotonics feature](https://github.com/rtic-rs/rtic/blob/acb9f446425a3bf6b00e542770bdbfd3c0bfe53e/examples/stm32f411_rtc_interrupt/Cargo.toml#L13) which I'm not seeing in the current code base.

Finomnis, 2024-04-29:
`rtic-monotonic` seems to be deprecated.

I'm sorry for the confusion, we had a full rtic-monotonics rewrite just a couple of weeks ago. (as @AfoHT pointed out, #874)

The rewrite is now merged and in the progress of being tested further before being released. It will probably happen soon.

---

> FYI: The `stm32f411_rtc_interrupt` example is also off, it references a yanked version of rtic that has an [rtic-monotonics [...]

Might be worth opening a separate issue for this.

closed 2024-06-17 by AfoHT