rtic-rs/rtic #938: Call to SCB::sys_reset() results in locked up core.

kareled, 2024-05-22:
Hello,
first of all, thanks a lot for the RTIC, it's amazing piece of code and technology!
I'm running it on stm32f4 and I'd like to enhance the application capability to be able to reset itself. Unfortunately simple:
```
cortex_m::peripheral::SCB::sys_reset();
```
results in following output from the probe-rs:
```
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
```
I've verified that if I just go without RTIC using stm32f4xx_hal & co, then sys_reset works like a charm. But not with RTIC. My RTIC app is around 15 user tasks and few hardware tasks and is using rtic monotonic.
Do you have any idea where should I look to debug this failure? Looks like I'm even not able to get hard fault frame for whatever reason...
Thanks!

glaeqen, 2024-05-22:
I assume you did try but it's not 100% clear to me from the phrasing: have you tried to make an empty RTIC application to reset itself or with all your user-code? Have you tried resetting your MCU by pulling the RST pin down and seeing if the behaviour is different?

kareled, 2024-05-22:
Indeed, this is an homework I still need to do. E.g. complex RTIC app runs on custom board without any support for RST pin while my "duplication" of sys_reset was done on nucleo board with comfortable RST button if needed.
Anyway, I'll make simple RTIC app for nucleo and will report back if sys_reset works or not there...

glaeqen, 2024-05-22:
Ok! FYI, RTIC has it's own official [Matrix channel](https://matrix.to/#/#rtic:matrix.org) where you can pop-in and ask questions to the wider audience if you want :)

closed 2025-03-12 by korken89