rtic-rs/rtic #939: rtic-monotonic on stm32 TIM20 or TIM17

govert-overgaauw-ampel, 2024-05-23:
Hi,

I am currently working on a micro-controller project running RTIC on STM32G4. We are using 21 timer channel outputs for PWM. So I i have not a lot of options for using RTIC monotonic except for TIM17 or TIM20. Systick is not an option I require some small delays, so the tick rate needs to be faster.

I tried implementing TIM20 based on the existing implementations, however TIM20 has seperate IRQs for TIM_UP and TIM_CC. This leads to a race condition that triggers the panic "monotonic must have missed an interrupt!". I am not well versed in all the nuances required to implement the monotonic timers. Is it possible with two hardware IRQs instead of one hardware timer IRQ? The abstraction layers only has a single on_interrupt call so both IRQs trigger the same function and check the flags.

TIM17 is another option, however this timer only has  a single capture compare, and the current implementations use two capture compares. One for the half period interrupt, and one is set by rtic-monotonics set_compare, enable/disable calls but I am not sure about the functionality and if this is required?

great-houk, 2024-07-06:
AFAIK this was fixed by #960, so it can be closed I think?

govert-overgaauw-ampel, 2024-08-09:
The issue with my implementation is that there are separate interrupts on the STM32G4 for TIM20 for TIM_UP and TIM_CC. So I was wondering if i can still use this implementation on that timer.

govert-overgaauw-ampel, 2024-09-10:
I worked around this potential issue by using a third capture compare as period interrupt on TIM20. This way I don't have to worry about potential race conditions between the two IRQs.

closed 2024-09-10 by govert-overgaauw-ampel