rtic-rs/rtic #944: Proplems while setting up rtic 2.1.1

Gospa123, 2024-05-29:
Helloo, 

I have been facing some troubles while trying to compile my code for the rtic 2.1.1 version. Right now I am just testing the sample code on migration that has been given on the site: https://rtic.rs/2/book/en/migration_v1_v2/complete_example.html. When trying to check/build I get an error that says no backend feature has been given, and also many other errors regarding the rtic-monotic and the modules, It says than it can't find the systick, module. The code is unchanged it is the same as on the page. Here is my Cargo.toml file.
```
[dependencies]
embedded-hal = "1.0.0"
cortex-m-rtic = "1.1.4"
systick-monotonic = "1.0.1"
panic-halt = "0.2"
cortex-m-semihosting = "0.5.0"
rtic = { version = "2.1.1", features = ["thumbv7-backend"] }
heapless = "0.8.0"
rtic-monotonic = "1.0.0"

[dependencies.stm32f3xx-hal]
features = ["stm32f303xc", "rt"]
version = "0.10.0"

# this lets you use `cargo fix`!
[[bin]]
name = "stm32f3-blinky"
test = false
bench = false

[profile.dev]
opt-level = 1
codegen-units = 16
debug = true
lto = false

[profile.release]
opt-level = "s"   # optimize for size
codegen-units = 1 # better optimizations
debug = true      # symbols are nice and they don't increase the size on Flash
lto = true        # better optimizations
```
Note: I am using stm32f3discovery board. If anyone could give me some guide how to build the project I would be grateful...

korken89, 2024-05-29:
Hi, you have mixed dependencies from v1 and v2.

Remove
- `systick-monotonic = "1.0.1"`
- `cortex-m-rtic = "1.1.4"`
- `rtic-monotonic = "1.0.0"`

Add
- `rtic-monotonics = { version = "2", features = ["cortex-m-systick"] }`

Gospa123, 2024-05-30:
Thank you a lot, it solved the problem.

closed 2024-06-17 by AfoHT