rtic-rs/rtic #948: Multicore Support

Quick-Flash, 2024-06-04:
After RTIC 0.5 multicore support was dropped. It is currently possible to sort of do multicore support by running 2 separate RTIC processes, one on each core, however this just means that each core is given its own tasks to do and has to communicate between each other in order to share data. New products such as the [ESP32P4](https://www.espressif.com/en/products/socs/esp32-p4) and the [rp2040 ](https://www.raspberrypi.com/documentation/microcontrollers/rp2040.html) are good candidates for an initial kind of multicore support as they have two identical cores. This way when scheduling new tasks whichever core isn't currently busy can be used instead of just running different tasks on each core. As more and more processors are being created that are multicore, this issue should become more important to better fully support these processors.

perlindgren, 2024-06-26:
We are experimenting with a new modular architecture for RTIC. The new "framework" suppors custom defined targets including multicores (both homogenous and heterogenous, both single binary and multi-bin).

Its still a POC but parts implemented works as intended. Its a complete re-write, and currently NOT syntax compatible 1-1 to current RTIC. There is nothing fundamental that hinders backwards compatibility, but at the moment effort has been put to simplicity. Thus the current syntax is (much) simpler. 

We can update this issue when we are at the point of public testing and requests for feedback.
/Per

Nashenas88, 2024-07-01:
@Quick-Flash sorry for adding noise to the bug, but do you have examples of how to get two rtic processes running on a multicore embedded image? I'm running multicore on rp2040 and really missing the ability to run an rtic app on the second core.

Quick-Flash, 2024-07-01:
> @Quick-Flash sorry for adding noise to the bug, but do you have examples of how to get two rtic processes running on a multicore embedded image? I'm running multicore on rp2040 and really missing the ability to run an rtic app on the second core.

Sadly no, I only know that is a possibility.

i-am-logger, 2024-11-28:
I like the notion of each core requires to run its own RTIC process...did anyone try to use DDS/micro-ros as a communication channel between RTIC processes?