rtic-rs/rtic #962: rtic-monotonics v2.0.2 not compiling

MikeGstefan, 2024-07-08:
rtic-monotonics v2.0.2 fails to compile with this error: 
` error[E0599]: no method named `fetch_add` found for struct `portable_atomic::AtomicU32` in the current scope `
at rtic-monotonics-2.0.2\src\systick.rs:106:25 and rtic-monotonics-2.0.2\src\systick.rs:126:25

This project has the following config: 
`
[build]
target = "thumbv6m-none-eabi"

[unstable]
build-std = ["core"]
`

korken89, 2024-08-07:
Hi @MikeGstefan 

Have you enabled the feature on `portable-atomic` to emulate CAS instructions for `thumbv6`?

optlink, 2024-09-15:
I just ran into this. I was able to hack around it by adding `portable-atomic` to my project and enabling `critical-section` since `rtic-monotonics` provides no way to enable this feature.

robamu, 2024-09-18:
I ran into this as well. Might be useful to add some docs for this?

robamu, 2024-09-18:
See https://github.com/rtic-rs/rtic/pull/977

closed 2024-10-02 by AfoHT