rtic-rs/rtic #974: rtt not attaching when using rtic on the rp2040

mxs2301, 2024-09-03:
When trying to run a simple hello world example with rtt on the raspberry pi pico H rtt fails to attach  to. 
I tried the same code with a microbit and it seemed to work just fine. 

Is there something that was wrongly configure here? 

Code for reference

```rs 
#![no_main]
#![no_std]

use rtic::app;

#[app(device = rp_pico::hal::pac, peripherals = true)]
mod app {
    use panic_probe as _;
    use rtt_target::{rtt_init_print, rprintln};

    #[local]
    struct Local {}

    #[shared]
    struct Shared {}

    #[init]
    fn init(mut cx: init::Context) -> (Shared, Local) {
        rtt_init_print!();
        rprintln!("Hello world");
        
        (Shared {}, Local {})
    }

    #[idle]
    fn idle(_cx: idle::Context) -> ! {
        loop {}
    }
}

```

mxs2301, 2024-09-09:
Found the issue, it was a simple case of skill issue on my side. 

The following code snippet was missing 

```rs
#[link_section = ".boot_loader"]
#[used]
static BOOT: [u8; 256] = rp2040_boot2::BOOT_LOADER_W25Q080;
```

I will close this issue and I am sorry for unnecessarily opening an issue.

Erhannis, 2024-10-24:
I had a problem that turned out to be the same, in the end, happening when I removed uses of `rp_pico`.  My bootloader declaration also had a `#[no_mangle]` line, though it doesn't seem to break when I remove it, so I don't know when it would matter.  Maybe something about optimizations.