rtic-rs/rtic #983: rtic_sync channels

maiteas, 2024-10-22:
Hi, I'm trying to implement RTIC code using channels, but I keep getting an error. Could you please share an example where two or more tasks communicate through channels? I can only find the example from the book at https://rtic.rs/2/book/en/by-example/channel.html. I based my implementation on that, but it didn't work, and I couldn't find any additional information online and I don't have any ideia how to make it works. Please help me! Thank you!

Erhannis, 2024-10-25:
I'm just passing through, but, what error are you getting?

AfoHT, 2024-10-30:
Have a look at https://github.com/rtic-rs/rtic/tree/master/examples/stm32f411_adc_and_mpsc_channel and the various `async-channel*` examples here: https://github.com/rtic-rs/rtic/blob/master/examples/lm3s6965/examples/async-channel-done.rs

And as @Erhannis kindly suggested, to get any help you need to provide something a bit more detailed, what you tried, results etc.!

maiteas, 2024-10-30:
Hello guys, thank you so much for your help. I really apreciate. Well, I did an example and it worked. But I have a question to ask. I'm sending a led state by this channel and it's not possible to see the state changing thought the led. Because it happens so fast. So my question is: can I use time function using async channels in RTIC?

Erhannis, 2024-11-24:
@maiteas To delay for a time, I use
```rust
use rtic_monotonics::rp2040::prelude::*; // There's probably similar things for different platforms
rp2040_timer_monotonic!(Mono);
...
async fn whatever_your_function_is(...) -> ... {
  ...
  Mono::delay(500_u64.millis()).await; // This bit's the important bit
  ...
}
```

closed 2025-03-12 by korken89