rtic-rs/rtic #985: RP2040 probe-rs no defmt/RTT, linker errors

Erhannis, 2024-10-23:
Ubuntu 24.04.1 LTS
RP2040 (Pico W)

For several hours I've been trying to get RTIC running (with defmt, and preferably debugging) on an RP2040.  The results are a little baffling.  I'm using probe-rs to flash the code.

I can compile and run https://github.com/rtic-rs/rtic/tree/master/examples/rp2040_local_i2c_init (if I add `runner = "probe-rs run --chip RP2040 --protocol swd"` to config.toml), but I get the warning "WARN probe_rs::util::rtt: No RTT header info was present in the ELF file. Does your firmware run RTT?"  I've tried adding defmt-rtt to Cargo.toml, added `"-C", "link-arg=-Tdefmt.x",` to config.toml, tried adding rtt features to various packages, but either nothing happens or I get linker errors.  (I did just discover I can get debugging to work on it, so that's a plus.)

I'm able to compile https://github.com/rtic-rs/defmt-app-template , once I fill in the various TODOs, but I get "ERROR probe_rs::cmd::run: Failed to attach to RTT, continuing...", and it doesn't seem like the code is actually running.  Also, somehow when I copy the main code from the i2c_init example over the main code of the template (and tweak the imports to work), I get a linker error about defmt???  I can copy the old file (containing defmt calls) right back over it, and the linking error goes away, which baffles me.  (I added some stuff to Cargo.toml to make the copied code compile, but haven't touched it since, and swapping the main code makes the linking error come and go, despite that both reference defmt.)  Linker error:
```
erhannis@mnode7:~/projects/test-app$ cargo run
   Compiling test-app v0.1.0 (/home/erhannis/projects/test-app)
error: linking with `flip-link` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/erhannis/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/erhannis/.wasmer/bin:/home/erhannis/pgms/mambaforge3/condabin:/home/erhannis/.sdkman/candidates/gradle/current/bin:/home/erhannis/pgms/flutter/bin:/home/erhannis/bin/:/usr/lib/qt6/bin/:/home/erhannis/.opam/default/bin:/home/erhannis/.cargo/bin:/home/erhannis/.local/bin:/home/erhannis/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/erhannis/.dotnet/tools:/home/erhannis/.local/share/coursier/bin" VSLANG="1033" "flip-link" "-flavor" "gnu" "/tmp/rustciNKspS/symbols.o" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/minimal-44dce56ff7d869f1.minimal.14834bf9c00790d0-cgu.0.rcgu.o" "--as-needed" "-Bstatic" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libpanic_probe-51d41fe991f3a057.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/librp2040_hal-1696bc028cd0826e.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libitertools-c855ee4f78a10994.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libeither-9a14dcb7a157a339.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libembedded_io-0e865585565e23a7.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/librand_core-e56e4353620c994b.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libbitfield-f6f9386f2683317a.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libusb_device-b6aee9dc5b468803.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libheapless-38d6f977d7718e54.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libhash32-b0b5e1436bbe58d5.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libbyteorder-0f93a574e5b7bc56.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libportable_atomic-59941480fd693a83.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libembedded_hal_nb-faa68e42fb8ccd19.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libpio-d0cb521148a2df6b.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libnum_enum-b2c2db6233258bc4.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libarrayvec-6c2d978685ed7040.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libfrunk-041ae04f04ccc508.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libfrunk_core-5e62c96d28c43a8f.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libembedded_dma-91c9f079a84ad61c.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libstable_deref_trait-bdb0c312334f72f2.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/librtic-1b286aaa9e02534e.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libatomic_polyfill-69f97e468f6f5417.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libbare_metal-ab854f731422c31e.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/librtic_core-2c825e754fea243a.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/librtic_monotonics-3c89667bc9965ac9.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/librp2040_pac-788714f8184c7338.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libcortex_m_rt-f6b85b8d60310977.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libcortex_m-cfcbf5cca1cc7863.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libembedded_hal-2b1b18ac0ab0cb11.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libvoid-85fe3fb7b05902af.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libnb-a88e4d1c73f85dfc.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libnb-19b9fdb772e1f2c0.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libvolatile_register-73613043dc7c5cc2.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libvcell-d880b71bacd066ff.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libbare_metal-d5a68899e31b9ad1.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/librtic_time-9da2a1160a89332c.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libembedded_hal_async-76fcdad864db69cd.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libembedded_hal-3fe5b323ff6ae444.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/librtic_common-3a89dd5a8509d6fd.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libfutures_util-af33c17acb8f1a68.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libpin_project_lite-74e020c4d44eef42.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libfutures_task-63cfb7bee31907c1.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libpin_utils-5a650b8bb6434b34.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libfutures_core-52b26d7e9d197894.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libcritical_section-754af05aeed63908.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libfugit-fab517c4f3263a90.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libgcd-629a4b9727d7fea8.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libdefmt-977c56ad642145cb.rlib" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/deps/libbitflags-f6aad9da853e795e.rlib" "/home/erhannis/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/librustc_std_workspace_core-3eaed70af7b15b94.rlib" "/home/erhannis/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/libcore-df1ccfea71d21752.rlib" "/home/erhannis/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/libcompiler_builtins-fbf8dd0c4df01e24.rlib" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/build/cortex-m-ab638b831eeb39ec/out" "-L" "/home/erhannis/projects/test-app/target/thumbv6m-none-eabi/debug/build/defmt- [... cut ...]
          >>> referenced 2 more times
          
          flip-link: the native linker failed to link the program normally; please check your project configuration and linker scripts
          

error: could not compile `test-app` (bin "minimal") due to 1 previous error
```

Another weird thing I noticed earlier - I was at some point able to flash code to the device via e.g. `cargo run` which didn't appear to run, but after I replugged the device it'd start running (e.g. led would start flashing).  It's not doing it now, though.

Erhannis, 2024-10-23:
While I'm not sure I've figured out all the weirdness that was going on, I did figure out some of it, and get a program running.  Ironically, it's about a behavior I noticed like 3 months ago and asked about on Stack Overflow, and then forgot.  https://stackoverflow.com/questions/78732542/rust-rp2040-entry-differences

I think at several points I removed the `rp_pico` crate, not needing it, and just using `rp2040_hal`.  However, as it turns out, `rp_pico` declares the bootloader.  If I didn't `use rp_pico` in some way, the bootloader didn't get written, and the code didn't run.  So adding this to the main rs file fixed it:
```
#[link_section = ".boot2"]
#[no_mangle]
#[used]
pub static BOOT2_FIRMWARE: [u8; 256] = rp2040_boot2::BOOT_LOADER_W25Q080;
```

I'm not sure what was going on with the linking errors, but unless they happen again I probably won't investigate further.

AfoHT, 2024-10-30:
I'm glad you got it running! :rocket:

I have not played enough with the rp2040 lately, so I do not have much to offer you, but regarding `boot2` I see the `local_i2c_init` example have this specified in the [memory.x](https://github.com/rtic-rs/rtic/blob/master/examples/rp2040_local_i2c_init/memory.x) file instead:

```
MEMORY {
    BOOT2 : ORIGIN = 0x10000000, LENGTH = 0x100
    FLASH : ORIGIN = 0x10000100, LENGTH = 2048K - 0x100
    RAM   : ORIGIN = 0x20000000, LENGTH = 256K
}

EXTERN(BOOT2_FIRMWARE)

SECTIONS {
    /* ### Boot loader */
    .boot2 ORIGIN(BOOT2) :
    {
        KEEP(*(.boot2));
    } > BOOT2
} INSERT BEFORE .text;
```

And it pulls in `rp-pico`, but this you already concluded was indeed a "must have".

If there's anything else, be it docs/examples etc. feel free opening this and continue in this issue, but for now I'll close it as it's not really actionable from RTIC PoV.

closed 2024-10-30 by AfoHT