rtic-rs/rtic #1001: Allow non literal integers to be used as a task priority

kareled, 2024-11-30:
Hello,

I'm attempting to get our application task priorities more under control and neither using enum type mapped to u8 nor u8 constants is working as a value to specify task priority. I always get an error that task macro requires integral literal and nothing more.

```
#[repr(u8)]
pub enum TaskPrio {
  Low = 1,
  Middle = 2,
  High = 3,
  Critical = 4,
[...]
  #[task(priority = TaskPrio::Low ....)
```
or
```
  const LOW : u8 = 1;
  const MIDDLE : u8 = 2;
  const HIGH : u8 = 3;
  const CRITICAL : u8 = 4;
[...]
  #[task(priority = LOW ....)
```
It would be great to have support for at least constants here if not enum type.

Tanks for considering that!
Karel

perlindgren, 2024-12-01:
In particular:
`rtic-macros`, provides the parser:

it looks something like this:

```rust
"priority" => {
                    if priority.is_some() {
                        return Err(parse::Error::new(
                            ident.span(),
                            "argument appears more than once",
                        ));
                    }

                    // #lit
                    let lit: LitInt = input.parse()?;

                    if !lit.suffix().is_empty() {
                        return Err(parse::Error::new(
                            lit.span(),
                            "this literal must be unsuffixed",
                        ));
                    }

                    let value = lit.base10_parse::<u8>().ok();
                    if value.is_none() {
                        return Err(parse::Error::new(
                            lit.span(),
                            "this literal must be in the range 0...255",
                        ));
                    }

                    prio_span = Some(lit.span());
                    priority = Some(value.unwrap());
                }
 ```
Important here is that `priority` (at this point) is declared as an `Option<u8>`, so the literal is parsed into a `u8`.

So the question here is how to turn structured data (even a named const) into a value  at compile time (or rather at proc-macro run-time). I don't think that's doable, but there might be some trick possible, not sure.

perlindgren, 2024-12-04:
We close this for now as not planned as a fix, but feel free to reopen if alternative solution can be foreseen.

closed 2024-12-04 by perlindgren