rtic-rs/rtic #1062: Add README to all crates

AfoHT, 2025-06-04:


mdabir1203, 2025-08-25:
Here's a starter take from mine. Would love to have a bit more direction what kind of README are you looking for 

# ondrop-utils

RAII helpers for running closures when values go out of scope.

This crate provides small but powerful utilities for **automatic cleanup** and **scope-based resource management** in Rust.  
It is designed with **no_std compatibility** in mind, making it suitable for embedded and safety-critical contexts where deterministic cleanup is essential.

---

## ✨ What’s Inside

- **[`OnDrop`]** — a scope guard that runs a `FnOnce` closure when dropped.  
- **[`OnDropWith`]** — wraps a value `T` and runs a `FnMut(&mut T)` closure on drop.  

These abstractions are useful for:
- Resource cleanup (files, sockets, buffers, locks)  
- Rolling back on failure (transactions, temporary states)  
- Embedded and safety-critical systems (ensures cleanup on every scope exit)  
- Enforcing invariants even in the presence of panics or early returns  

---

## 🚀 Installation

Add this to your `Cargo.toml`:

```toml
[dependencies]
ondrop-utils = "0.1"

AfoHT, 2025-09-17:
Hi, thank you for showing interest!

As you can tell there's a lot of prior art for the various crates in this repo.

This ticket is a TODO for going through all crates and making sure they all

- Got a README
- README should match in style and content the other READMEs in the repo
- Check links and current information in existing READMEs

See #1074 for inspiration