rtic-rs/rtic #1088: Support for shared resources in exception handlers on ARMv6-M

FeldrinH, 2025-07-16:
According to the RTIC documentation ([source](https://rtic.rs/2/book/en/internals/targets.html)):
> [...] ARMv6-M/ARMv8-M-base cannot have tasks with shared resources bound to exception handlers, as these cannot be masked in hardware.

This sentence seems to indicate that on ARMv6-M something like this won't work (and indeed on RTIC v1 this produced a compile-time error):
```rust
#[task(binds = SysTick, shared = [peripheral])]
fn watchdog(mut ctx: watchdog::Context) {
    ctx.shared.peripheral.lock(|p| p.whatever());
}
```

However, with RTIC v2 this code compiles just fine on `thumbv6m-none-eabi` with the `thumbv6-backend` feature enabled.

What's going on here? Is there a bug in RTIC? Is there outdated info in the documentation? Am I just completely misunderstanding something?

FeldrinH, 2025-08-14:
This seems rather troublesome. If I understand correctly the current behavior could lead to UB in safe code. Is there hope for a quick fix?

FeldrinH, 2025-08-14:
A simple but somewhat heavy-handed fix might be to just use critical sections when one of the handlers using a shared resource is an exception handler.

FeldrinH, 2025-08-19:
> A simple but somewhat heavy-handed fix might be to just use a global critical section when one of the tasks using a shared resource is an exception handler.

I made a PR implementing this fix. Trying to selectively disable specific exceptions seemed quite convoluted, so I left that as a future development for someone with more time and insight.

perlindgren, 2025-08-20:
If I recall correctly, changing the priority of an executing interrupt handler may lead to unexpected/undefined behavior on v6m, and as pointed out in the PR the idle task has no running priority, so that approach is not really applicable.

Disabling individual exceptions should work but each exception need some special treatment. For non-maskable exceptions (NMI) the integrity of shared resources might be compromised (as an NMI might occur at any time), so if we want to ensure soundness (guaranteed race-free execution) I don't think there is a way.

One observation though. As long as an exception bound task(s) has the same priority as all other (interrupt bound) tasks accessing the same set (or subset) of resources one should be able to use lock free implementation for both the exception handler(s) and corresponding interrupt driven task(s), but that is really an edge case. 

I don't see an elegant solution for this problem, brute force exception masking seems to be the only way (or using the FAULT_MASK to disable all maskable exceptions) may be the least bad alternatives (due to limitations in the v6m interrupt/exception hardware). Using FAULT_MASK will affect *only* maskable exceptions, so its not really a global critical section. The lock code generation for normal tasks should disable other normal tasks as we do now, AND set the FAULT_MASK if shard with any exception handler. Codegen for exceptions need to disable other normal tasks as we do now, AND set the FAULT mask  in case there is any other (maskable) exception that shares the same resource AND has higher priority. I hope/think that should cover all cases.
/Per

FeldrinH, 2025-08-20:
It seems that all the options have their own issues. 

> We should just emit a compilation error in case of exception handlers with shared resources (like in RTIC-v1).

Worth noting that adding this compilation error now would be a breaking change for everyone that is currently using shared resources in exception handlers.

FeldrinH, 2025-12-11:
Question: Would it be possible to fix the locking by dynamically lowering the priority of exceptions to mask them out? It seems like that should work. If I were to implement that and make a PR, would you merge it?